Mystfit/ESP32-BLE-CompositeHID #2: Only recognized by steam input

dtomkovicz, 2024-04-19:
Hello, I'm having trouble getting the device recognized in unity as a game controller, it always appears as a joystick. This is the case with my custom code, and when simply using the xinput example file. Is there something the example is missing to display as a normal gamepad?

Steam input recognizes the controller and even sends my debug vibration callback to make my esp32 output serial.

Mystfit, 2024-04-25:
Most of testing has been with Steam in order to make sure that it recognizes the controller as being an XBox controller using the BLE XInput driver. If Unity is not recognizing it as a XInput gamepad, they may using their own detection system that isn't picking up the controller properly and is instead falling back to a generic joystick with axes/buttons. I'm assuming that you're using the XboxXInputController example sketch?

dtomkovicz, 2024-04-27:
> Most of testing has been with Steam in order to make sure that it recognizes the controller as being an XBox controller using the BLE XInput driver. If Unity is not recognizing it as a XInput [...]

Yup! That's the example sketch I was referring to. It also seems to not show as a standard gamepad on most gamepad tester websites I found. I've done some reading on steam input and it seems like they shave more flexible mapping layers to so my best guess is it basically only works with steam input (at least only seems to 100% work there, rumble and other things I can't get to work on any tester site).

I honestly have no idea how to code HID descriptors or anything like that so I'm very appreciate it works at all because this was the only library I could find to get a Bluetooth gamepad with force feedback up and running, but still wanted to reach out about it in case there are slight adjustments I could make to the script or library to get it to be picked up - unsure. In the short term I'm looking into implementing steam input into Unity but am still waiting on valve to approve my dev profile so I get an app ID.

Mystfit, 2024-04-27:
Something that we also could try is renaming the Bluetooth device to a
default XBox One S controller name just in case Unity is checking device
names.

Could you post some of the gameplay testing websites you used here so I
could verify for myself? Also, would you mind sharing your script or at
least the gamepad initialization code here as well so I could take a look?

Mystfit, 2024-05-05:
@dtomkovicz I've done some testing with the controller and Unity. I used my XboxInputController.ino test sketch but I used the `XboxSeriesXControllerDeviceConfiguration` config instead of the One S config (see line 29/30).

In Unity, I checked the value of `UnityEngine.InputSystem.Gamepad.current` and it returned `XInputControllerWindows:/XInputControllerWindows` which is an XInput gamepad instead of a joystick (I think).

Does this help?

dtomkovicz, 2024-05-05:
Sorry I didn't get back to you on the first message, I've been away on a work trip for a bit! I will send a long with websites when I get the chance. I did try swapping to the other config I believe but will certainly double check if it seemingly worked for you! Did you also change the manufacturer/product  name etc?

I really appreciate you trouble shooting this.

dtomkovicz, 2024-05-15:
Please forgive all my delays on this, this is the gamepad tester website I normally use, I've attached images of what this looks like on my xbox one controllers (looks like they have a different verdor ID but steam still shows them as "Xbox One S Controller" so I am unsure the distinction between the one you are emulating (which I assume must be different given your code comments and the different product ID.
[image]
vs
[image]

In unity, I see the xbox controller in the input debugger like this:
[image]
vs. (my script & ex. script):
[image]

I double checked both X and S and in both cases the tester and unity don't see it as a controller but a joystick, but steam input does.

The following websites somewhat work where some inputs are recognized, but seemingly not all (for example the d pad is not):
https://www.onlinemictest.com/controller-tester/
https://greggman.github.io/html5-gamepad-test/
https://gamepadtester.net/

In the first two of these websites I see my actual xbox controller as a standard gamepad while the other ones is unrecognized.
[image]

Unsure if this has to do with the values inthe HID descriptor that can be seen in the unity input debugger I linked.

dtomkovicz, 2024-05-15:
> @dtomkovicz I've done some testing with the controller and Unity. I used my XboxInputController.ino test sketch but I used the `XboxSeriesXControllerDeviceConfiguration` config instead of the One S [...]

I should also add I get null when logging this, and with the xbox series X config uncommented and the other commented my debug output looks like this:

[image]

Mystfit, 2024-05-16:
Thanks for the extensive testing!

Regarding Unity, I just fired up the standard XboxXInputController.ino sketch with that input debugger and was able to get the SeriesX config to be recognized by the input debugger with no modifications. This was tested this on Windows 11 version 23H2 (OS Build 22631.3593).
[image]

However, Unity complained about the OneS configuration and showed the device description as: 
```
{
    "interface": "HID",
    "type": "",
    "product": "107135773536",
    "serial": "107135773536",
    "version": "1032",
    "manufacturer": "ੲ",
    "capabilities": "{\"vendorId\":1118,\"productId\":765,\"usage\":5,\"usagePage\":1,\"inputReportSize\":17,\"outputReportSize\":9,\"featureReportSize\":0}"
}
```

For comparison, this is the device description from the SeriesX config:
```
{
    "interface": "XInput",
    "type": "",
    "product": "",
    "serial": "",
    "version": "",
    "manufacturer": "",
    "capabilities": "{\"userIndex\":0,\"type\":1,\"subType\":1,\"flags\":6,\"gamepad\":{\"buttons\":62463,\"leftTrigger\":255,\"rightTrigger\":255,\"leftStickX\":-32768,\"leftStickY\":-32768,\"rightStickX\":-32768,\"rightStickY\":-32768},\"vibration\":{\"leftMotor\":65535,\"rightMotor\":65535}}"
}
```

I'll look into creating another config to match the Xbox Wireless controller product ID to see if that helps.

dtomkovicz, 2024-05-16:
Trouble shooting on a friend's pc revealed that the controller was recognized there (using series x config like you said!), so I believe I may have some bad drivers installed or something and am trying to see if I can wipe out what's blocking me without having to reset my OS or something. I'm very encouraged by this as it narrows the scope for at least my series x config issue to be unique to my pc. My first guess were these nefarious drivers (dev name not malware) I had installed to use a service called dual senseX which I had been using for some prototyping. Uninstalling that didn't make a difference so far so I am still looking, very interesting that whatever is interfering with my pc is affecting your device code but seemingly has no impact on any of my other controllers, including my other Xbox one s oem controllers.

I'll be doing more testing later tonight to see if I can get the series x config working on my pc!

dtomkovicz, 2024-05-17:
I was able to get it working! It required making my PC forget the bluetooth device completely also before it worked after switching which I didn't realize, might even be the only thing necessary I'm honestly not sure. I can keep the issue open for the other config I suppose? Right now I think everything is working for the needs of my prototype though so I'm thrilled! I appreciate you looking into this even though the fix was unrelated to your code. 

It also seems like the stick click isn't registering on this gamepad website , but it is in Unity and steam input.

https://hardwaretester.com/gamepad

Mystfit, 2024-05-17:
Ah, yes now I remember. I usually remove the bluetooth device if I modified the HID descriptor, PID or VID since I think that the computer caches these values. I'm assuming in non-development scenarios these values are unlikely to change which could be the cause of this behaviour.

I just tested the SeriesX config on that website and B11 and B12 seemed to light up when the left and right sticks were pressed in. I also tested it with Unity and leftStickPress and rightStickPress worked in the input debugger.

I'm going to close this issue as the initial question has been resolved, but I'll move the OneS HID descriptor problem to another issue.

closed 2024-05-17 by Mystfit