Mystfit/ESP32-BLE-CompositeHID #4: How do I program a mouse click?

Knockoi, 2024-04-24:
I want to control the mouser pointer with the joystick, but I don't know how to give the command to click the left button?

Knockoi, 2024-04-25:
mouse->press(MOUSE_LEFT);
 mouse->release(MOUSE_LEFT);

Mystfit, 2024-04-29:
Apologies, seems as though I forgot to include an example for mouse presses.

The function is `mouse->mousePress(MOUSE_BUTTON)` or `mouse->mouseRelease(MOUSE_BUTTON)` where `MOUSE_BUTTON` is one of:

- MOUSE_LOGICAL_LEFT_BUTTON
- MOUSE_LOGICAL_RIGHT_BUTTON
- MOUSE_LOGICAL_BUTTON_3
- MOUSE_LOGICAL_BUTTON_4
- MOUSE_LOGICAL_BUTTON_5

For example, the following will press the left mouse button for 100ms then release it.

```
mouse->mousePress(MOUSE_LOGICAL_LEFT_BUTTON);
mouse->sendMouseReport();
delay(100);
mouse->mouseRelease(MOUSE_LOGICAL_LEFT_BUTTON);
mouse->sendMouseReport();

```

Knockoi, 2024-04-30:
Thank you very much!!! 
For the example you provided, what about the Keyboard example? I want to press the left arrow and the right arrow, how do I write it?
How can I check it in the future?

closed 2024-05-01 by Knockoi