Mystfit/ESP32-BLE-CompositeHID #5: Missing LIBARY "Callback.h"

Knockoi, 2024-04-25:
In file included from c:\Users\2006e\OneDrive\文件\Arduino\libraries\ESP32-BLE-CompositeHID\GamepadDevice.cpp:1:
c:\Users\2006e\OneDrive\文件\Arduino\libraries\ESP32-BLE-CompositeHID\GamepadDevice.h:7:10: fatal error: Callback.h: No such file or directory
 #include <Callback.h>
          ^~~~~~~~~~~~
compilation terminated.
exit status 1

Compilation error: exit status 1

Mystfit, 2024-04-25:
Oops! Looks as though I've missed a library dependency. Install the [Callback](https://www.arduino.cc/reference/en/libraries/callback/) library through the Arduino library manager.

Knockoi, 2024-04-25:
Thank you so much for getting back to me!!!!
You are my superman and you saved my topic!!!!

closed 2024-04-25 by Knockoi