Mystfit/ESP32-BLE-CompositeHID #7: Can you make an example of how to reset or re-pair Bluetooth ? PLZ

Knockoi, 2024-06-13:
I like very much the ESP32-BLE-CompositeHID bluetooth, but I would like to be able to order more things, such as: re-pairing, and auto-connecting previously connected devices, etc. I hope you can expand this libary, bye!!!!

Mystfit, 2024-10-19:
Thanks Knockoi! That's good feature request, I'll look into exposing that functionality from NimBLE. At the moment, you should be able to pair/re-pair to the device by removing from your Bluetooth devices window (depending on which OS you're using) and adding it again. The device should also re-connect as long as it is a paired and powered on.

Roet-Ivar, 2024-11-14:
Is there a way to simulate holding the sync button (I know it sounds stupid), BlueRetro won't pair up for some reason?

Knockoi, 2024-11-15:
It's like storing connected devices, right?