Mystfit/ESP32-BLE-CompositeHID #10: Changing Name

ernest-cpu, 2024-07-22:
[image]
I am not able to change the name of the game controller im making. It automatically sets itself as however number of inputs I picked.

NafisAhnnaf, 2024-08-13:
Also facing the same issue. And turns out the button mappings are not okay. The example of XBOX gamepad is not working appropriately. Also the rumble implementation code does not work either. It needs a few more revisions

Mystfit, 2024-10-19:
Thanks for the feedback. 

@DrSimgear At the moment we can only set the name of the Bluetooth device, not the actual HID device which is why in the Game Controllers window the name is generic. The Xbox controller example is different since the name is controlled by the PID and VID that we set. However, I will research this further to see if it is possible to set the name of the gamepad.

@NafisAhnnaf I've tracked down some potential issues that may have cropped up in the latest NimBLE library that may have broken the XBox controller emulation. If you're using NimBLE 1.4.2, then the PID and VID advertisements will be incorrect, so Windows won't correctly identify the controller as an XBox controller and won't use the correct BLE XInput driver. This causes things like the button assignments to be incorrect and the rumble to not work, since for an XBox controller, this all needs to go through XInput.

I've updated the installation instructions with how to install the [latest Nimble-Arduino library (main branch)](https://github.com/h2zero/NimBLE-Arduino) that contains the fixes that we need and I've also updated this library to fix the PID and VID issues so the XBox gamepad example should work again. 

I just tested the XboxXInputController.ino example sketch with the latest Nimble-Arduino library (see above) and was correctly identified by Windows and Steam. Rumble is also working when using the Steam controller test panel, clicking the rumble button in steam turned the LED on and off.
[image]

Mystfit, 2024-10-19:
@DrSimgear Found the [following ](https://github.com/afpineda/OpenSourceSimWheelESP32/blob/main/doc/RenameDeviceWin_en.md#why-all-this-mess ) which goes into detail about where the HID device name is retrieved from on Windows. In summary, it looks as though we can't set the name of the HID device for BLE devices unless you modify the registry. You could potentially set a custom PID/VID for each device you have and then modify the registry to change the names linked to each PID/VID but that strikes me as being quite an unfriendly process for the end-user unless you're running some app on the destination computer that can fiddle around with the registry for you.

I'm going to close this for the moment unless we can find an alternative solution for BLE.

closed 2024-10-19 by Mystfit

DrSimgear, 2024-10-19:
to get around this issue I simply set the vid and pid of the usb and bluetooth the same on my controller. by connecting the usb a single time it sets the name in registry and if you ever connect the controller through bluetooth then it will receive the correct name. The alternative is to use bluetooth classic.

Mystfit, 2024-10-19:
That's a clever solution. What happens when you connect the BLE first and then connect the USB? Does it use the old cached generic name or does it use the USB HID device name?

DrSimgear, 2024-10-20:
Once you connect the usb once, windows will remember the name forever unless you delete it from registry. If you connect through bluetooth and windows doesn't have a name for the hid device then it will make one up temporarily based on how many buttons and axis your device has. If you ever plug in the usb instead then it will save that name and next time bluetooth connects it will use the name it saved instead of making one up. There is a issue with the bluetooth vid/pid on ble, for some reason it is backwards so you have to write it backwards for windows to see them as the same hid device. Also for ble iirc the buttons don't start at the same number as usb hid, I don't remember if it was 0 or 1 so you have to compensate and add 1 to the hid that starts at 0. another issue is if you connect the usb while ble is already connected then you will have two devices with the same name, you have to follow the proper procedure to turn off ble with nimble or just disconnect the device and turn off advertising. If you just turn of ble then next time you turn it on the memory will overflow because you didn't clear the previous ble instance. I couldn't fix this because i'm not familiar with c++ syntax to be able to directly use nimble functions.