Mystfit/ESP32-BLE-CompositeHID #16: Compatibility with BlueRetro

Roet-Ivar, 2024-11-14:
For some reason I can't pair up with BlueRetro, a PC works fine, but it seems like it expect an xbox one controller to behave slight differently and I can't figure out why.

Mystfit, 2024-11-16:
Can you confirm which version of NimBLE-Arduino are you using and are you using the version of ESP32-BLE-CompositeHID? There were some fixes made which should have fixed incorrect PID and VID values.

Also, are you basing your custom controller on the [XboxXInputController.ino](https://github.com/Mystfit/ESP32-BLE-CompositeHID/blob/master/examples/GamepadExamples/XboxXInputController/XboxXInputController.ino) sketch or a generic controller? I've not used BlueRetro myself so I'll have to look up how it works.

Roet-Ivar, 2024-11-17:
> Can you confirm which version of NimBLE-Arduino are you using and are you using the version of ESP32-BLE-CompositeHID? There were some fixes made which should have fixed incorrect PID and VID [...]

Hi! I've based on the Xbox Input Controller using Nimble from the commit you linked in the description. Blueretro is very cool, even cooler in conjuction with something like this ❤️

Mystfit, 2024-11-18:
@Roet-Ivar Ok, I just looked up the XBox pairing guide. Seems as though the BlueRetro is waiting for when an XBox controller is put into pairing mode. Since by default CompositeHID is always advertising, I wonder the BlueRetro if it is trying to bond with the ESP32 in a way that I'm not familiar with. I think that it verifies what device it is connected to via HID descriptors, so that should generally be enough for it to detect the device as an XBox controller since I'm using the same HID descriptors. Are you receiving any sort of indication from the BlueRetro that it can see the existence of your ESP32 running the XboxXInputController.ino sketch?

Interestingly enough, there's a [discussion](https://github.com/darthcloud/BlueRetro/discussions/311) about pairing with ESP32-BLE-Gamepad, which is what I based my BLE code around.

Roet-Ivar, 2024-11-18:
@Mystfit I was just about to make some more pairing tests (tonight hopefully), realized I was on an older firmware. Let me come back to you when tested with the latest FW to see that this is still an issue, don't wanna waste your time :)

Mystfit, 2024-11-21:
@Roet-Ivar How did it go. Any luck?

Roet-Ivar, 2024-11-21:
@Mystfit Hi, sorry for my late reply! I've retried now making sure using Nimble from the commit linked. Unfortunately no success.

Roet-Ivar, 2024-12-09:
I'm just writing my findings here just in case it might help.
The Composite-HID-device attempts to auto sync with the BlueRetro when powered on, the BlueRetro picks this up and stops it's LED blinking (waiting for device to pair), eventually the Composite-HID-device disconnects and the BlueRetro is back to blinking again. 

When paring with firmware from the original ESP32-BLE-Gamepad repo the also auto-connects, but successfully this time. 

What could be the difference in advertising it's pairing routine between yours and the original implementation?

Also, I understand if this is not the most interesting issue for you to look into, I just found your implementation more intriguing due to that it's centered around a device that is well known and used.

LeeNX, 2025-01-05:
Interesting issue, not that I can help with it per se, but maybe look at using the PlatformIO setup. I have found it way easier to pin library decencies per project vs globally installed. Especially useful when testing  and debugging different versions of libraries. I have not been so successful with Arduino IDE.

davidxgames, 2025-02-27:
> For some reason I can't pair up with BlueRetro, a PC works fine, but it seems like it expect an xbox one controller to behave slight differently and I can't figure out why.

Hello everyone, I've found the solution to the BlueRetro connection issue. I'm preparing a pull request to implement it. Although the change is minimal, it was enough to prevent BlueRetro from syncing correctly.

Roet-Ivar, 2025-02-27:
These are great news!

Roet-Ivar, 2025-02-27:
> Hello everyone, I've found the solution to the BlueRetro connection issue. I'm preparing a pull request to implement it. Although the change is minimal, it was enough to prevent BlueRetro from [...]

This is great!
Please keep us updated

davidxgames, 2025-02-27:
> This is great! Please keep us updated

You need to change line 280 in BleCompositeHID.cpp to pAdvertising->setAppearance(HID_GAMEPAD);. I managed to solve the connectivity with Windows—it reconnects automatically just like with bluieretro—but I think they have already fixed that.

Roet-Ivar, 2025-02-28:
> You need to change line 280 in BleCompositeHID.cpp to pAdvertising->setAppearance(HID_GAMEPAD);. I managed to solve the connectivity with Windows—it reconnects automatically just like with [...]

Thank you so much for sharing, PR incoming? 
I'll try this during the weekend!

Mystfit, 2025-02-28:
Good catch, I'll test this during the weekend. I'll also look into exposing
HID_GAMEPAD as a configurable option.

davidxgames, 2025-03-01:
> Good catch, I'll test this during the weekend. I'll also look into exposing
> HID_GAMEPAD as a configurable option.
> […](#)

I noticed that on Windows the device was recognized as an Xbox controller, but the icon displayed was that of a keyboard, which I wasn’t happy with. So I looked into how to change it, and once I managed to do so, I retested with BlueRetro and it connected properly. Additionally, I encountered an issue with the rumble feature in BlueRetro, as it always returned values of 0, from the moment of connection, the callback was continuously being sent, and when I got hit in the game, I kept receiving 0 for the motor values. Consequently, I took the time to analyze the raw callback, and that’s how I managed to implement the rumble function on both Windows and BlueRetro.

Roet-Ivar, 2025-03-05:
Great job everybody! 
I'm going to write an implementation for my esp32-c3 Famicom Controllers this weekend.