Mystfit/ESP32-BLE-CompositeHID #18: [Feature request] Compatibility with gyroscope.

shivamkantprasad, 2024-11-21:
I wanna make a controller with motion control like in PS controllers. I want to have an option to send gyroscope data to pc like some controllers send. It would be greatly appreciated if you could add this feature.

lemmingDev, 2024-11-23:
Would be trivial to simply manually map gyroscope data to left analog x and y etc...

Mystfit, 2024-11-23:
I agree with what @lemmingDev suggessted, mapping the gyro to existing axis inputs would be the easiest way to accomplish this as then it would be compatible with all games that use analog joysticks. I've previously done something similar by mapping a hall effect magnetic sensor to triggers.

The only controllers I can think of that pass dedicated gyro information along to a PC are the Dualsense controllers. I think they're compatible with Steam natively, but you'd still have to use Steam Input to map the gyro to standard controller axes.

Can you provide some examples of gyroscopic game controllers that send data to the PC and any games in question I can look at as an example?

shivamkantprasad, 2024-11-29:
switch pro controller does send data to pc and many switch emulator can use that data to do stuff.

shivamkantprasad, 2024-11-29:
> Would be trivial to simply manually map gyroscope data to left analog x and y etc...

well if I would do that that will not let me also have two analog sticks

Mystfit, 2024-11-29:
> switch pro controller does send data to pc and many switch emulator can use that data to do stuff.

It sounds as though we're specifically talking about Switch Pro controllers then. I did find a [HID descriptor](https://gist.github.com/ToadKing/b883a8ccfa26adcc6ba9905e75aeb4f2) for a pro controller that we could use as a starting point and we could extend the XBoxGameController class using this HID descriptor instead.

shivamkantprasad, 2025-01-20:
also, PS controller have gyroscope too that works with the help of [DS4Windows](https://ds4-windows.com/) [(Github)](https://github.com/Ryochan7/DS4Windows). This can help people hook up emulators like [ryujinx](https://ryujinx-emulator.com/) with there custom made controllers. (ps. I am not so good at coding to make my own changes to this code according to the hid descriptors you have sent above).

With the GitHub link of DS4 I think I should be easier to emulate a PS controller as they maybe have done reverse engineering.