Mystfit/ESP32-BLE-CompositeHID #19: Axis 2 & 3 set to 1 always

maxr217, 2024-12-09:
First of all thank you for porting the NimBLE to Xinput HID. 

The latest version of NimBLE has an issue where axis 2 and 3 are always set to max & the set right joystick command does not work. I believe the work around is to set the right joystick values z and rZ independently. I just wanted to raise awareness that this is currently seen in this branch as well.

felixstorm, 2024-12-28:
@maxr217 The problem seems due to the whole memory block being initialized as `0x00` [here](https://github.com/Mystfit/ESP32-BLE-CompositeHID/blob/39838f1d5057121c198b4a96cf03bec3f3a9d80c/XboxGamepadDevice.cpp#L103) but for the center the axis values should actually be `0x8000`. And just setting them to `0, 0` does not work either due to a problem with the comparison for changed axis values (raw values being compared to the values in the `_inputReport` which already have the offset added [here](https://github.com/Mystfit/ESP32-BLE-CompositeHID/blob/39838f1d5057121c198b4a96cf03bec3f3a9d80c/XboxGamepadDevice.cpp#L149)).

As a workaround, this should do the trick:
```cpp
gamepad->setLeftThumb(1, 1);
gamepad->setLeftThumb(0, 0);
gamepad->setRightThumb(1, 1);
gamepad->setRightThumb(0, 0);
```

Mystfit, 2025-03-08:
@maxr217 Thanks for noticing this issue! 

@felixstorm Good catch on the memset `0x00` values not matching what we expect the actual values to be with the center offset `0x8000`. I'll make sure that when we are resetting the device values that we center all axes.

closed 2025-03-08 by Mystfit