Mystfit/ESP32-BLE-CompositeHID #21: Nimble problems

Laurin08, 2024-12-23:
I get several errors while compiling the example gamepad code. I think the problem is the NimBLE library. Which version do you guys use? With the older version (2.0.0) I got less errors but still something:

/Users/laurinmichel/Documents/Arduino/libraries/ESP32-BLE-CompositeHID/GamepadDevice.cpp: In member function 'uint8_t GamepadDevice::specialButtonBitPosition(uint8_t)':
/Users/laurinmichel/Documents/Arduino/libraries/ESP32-BLE-CompositeHID/GamepadDevice.cpp:343:20: error: 'invalid_argument' is not a member of 'std'
343 | throw std::invalid_argument("Index out of range");
| ^~~~~~~~~~~~~~~~
/Users/laurinmichel/Documents/Arduino/libraries/ESP32-BLE-CompositeHID/GamepadDevice.cpp:6:1: note: 'std::invalid_argument' is defined in header ''; did you forget to '#include '?
5 | #include "esp32-hal-log.h"
+++ |+#include
6 | #define LOG_TAG "GamepadDevice"

exit status 1

Compilation error: exit status 1

cflixdev-projects, 2024-12-23:
You need to download these commit versions but before u should delete them from Arduino Library Manager:
## Dependencies
 - Nimble-Arduino. Use the latest Github version from commit [a79941c](https://github.com/h2zero/NimBLE-Arduino/commit/a79941cc9e77ed60136ee385b1c73d89c847bd94) onwards. Do not use 1.4.2 from the Arduino library manager.
 -  Callback. Available from the Arduino library manager or [here](https://github.com/tomstewart89/Callback).

Laurin08, 2024-12-23:
Hey 
Thanks for your answer but I already installed the Callback library and the new Nimble version has commit [a79941c](https://github.com/h2zero/NimBLE-Arduino/commit/a79941cc9e77ed60136ee385b1c73d89c847bd94) already integrated, if I saw that right.
if I use the newest nimble version (2.1.2) I get even more errors. 
I don't really know were the issue could be. I also tried it on another laptop where only the needed libraries were installed.

the errors with version 2.1.2:

```
/Users/laurinmichel/Documents/Arduino/libraries/ESP32-BLE-CompositeHID/GamepadDevice.cpp: In member function 'virtual void GamepadDevice::init(NimBLEHIDDevice*)':
/Users/laurinmichel/Documents/Arduino/libraries/ESP32-BLE-CompositeHID/GamepadDevice.cpp:147:23: error: 'class NimBLEHIDDevice' has no member named 'inputReport'; did you mean 'getInputReport'?
  147 |     auto input = hid->inputReport(_config.getReportId());
      |                       ^~~~~~~~~~~
      |                       getInputReport
/Users/laurinmichel/Documents/Arduino/libraries/ESP32-BLE-CompositeHID/GamepadDevice.cpp:150:24: error: 'class NimBLEHIDDevice' has no member named 'outputReport'; did you mean 'getOutputReport'?
  150 |     auto output = hid->outputReport(_config.getReportId());
      |                        ^~~~~~~~~~~~
      |                        getOutputReport
/Users/laurinmichel/Documents/Arduino/libraries/ESP32-BLE-CompositeHID/GamepadDevice.cpp: In member function 'uint8_t GamepadDevice::specialButtonBitPosition(uint8_t)':
/Users/laurinmichel/Documents/Arduino/libraries/ESP32-BLE-CompositeHID/GamepadDevice.cpp:343:20: error: 'invalid_argument' is not a member of 'std'
  343 |         throw std::invalid_argument("Index out of range");
      |                    ^~~~~~~~~~~~~~~~
/Users/laurinmichel/Documents/Arduino/libraries/ESP32-BLE-CompositeHID/GamepadDevice.cpp:6:1: note: 'std::invalid_argument' is defined in header '<stdexcept>'; did you forget to '#include <stdexcept>'?
    5 | #include "esp32-hal-log.h"
  +++ |+#include <stdexcept>
    6 | #define LOG_TAG "GamepadDevice"
/Users/laurinmichel/Documents/Arduino/libraries/ESP32-BLE-CompositeHID/BleConnectionStatus.cpp: In member function 'virtual void BleConnectionStatus::onConnect(NimBLEServer*, NimBLEConnInfo&)':
/Users/laurinmichel/Documents/Arduino/libraries/ESP32-BLE-CompositeHID/BleConnectionStatus.cpp:9:31: error: invalid use of incomplete type 'class NimBLEConnInfo'
    9 |     pServer->updateConnParams(connInfo.getConnHandle(), 6, 7, 0, 600);
      |                               ^~~~~~~~
In file included from /Users/laurinmichel/Documents/Arduino/libraries/ESP32-BLE-CompositeHID/BleConnectionStatus.h:9,
                 from /Users/laurinmichel/Documents/Arduino/libraries/ESP32-BLE-CompositeHID/BleConnectionStatus.cpp:1:
/Users/laurinmichel/Documents/Arduino/libraries/NimBLE-Arduino/src/NimBLEServer.h:44:7: note: forward declaration of 'class NimBLEConnInfo'
   44 | class NimBLEConnInfo;
      |       ^~~~~~~~~~~~~~
/Users/laurinmichel/Documents/Arduino/libraries/ESP32-BLE-CompositeHID/BleCompositeHID.cpp: In member function 'void BleCompositeHID::setBatteryLevel(uint8_t)':
/Users/laurinmichel/Documents/Arduino/libraries/ESP32-BLE-CompositeHID/BleCompositeHID.cpp:147:25: error: 'class NimBLEHIDDevice' has no member named 'batteryLevel'; did you mean 'setBatteryLevel'?
  147 |             this->_hid->batteryLevel()->notify();
      |                         ^~~~~~~~~~~~
      |                         setBatteryLevel
/Users/laurinmichel/Documents/Arduino/libraries/ESP32-BLE-CompositeHID/BleCompositeHID.cpp: In static member function 'static void BleCompositeHID::taskServer(void*)':
/Users/laurinmichel/Documents/Arduino/libraries/ESP32-BLE-CompositeHID/BleCompositeHID.cpp:221:36: error: 'class NimBLEHIDDevice' has no member named 'reportMap'; did you mean 'setReportMap'?
  221 |     BleCompositeHIDInstance->_hid->reportMap(&customHidReportDescriptor[0], hidReportDescriptorSize);
      |                                    ^~~~~~~~~
      |                                    setReportMap
/Users/laurinmichel/Documents/Arduino/libraries/ESP32-BLE-CompositeHID/BleCompositeHID.cpp:224:36: error: 'class NimBLEHIDDevice' has no member named 'manufacturer'; did you mean 'setManufacturer'?
  224 |     BleCompositeHIDInstance->_hid->manufacturer()->setValue(BleCompositeHIDInstance->deviceManufacturer);
      |                                    ^~~~~~~~~~~~
      |                                    setManufacturer
/Users/laurinmichel/Documents/Arduino/libraries/ESP32-BLE-CompositeHID/BleCompositeHID.cpp:267:36: error: 'class NimBLEHIDDevice' has no member named 'pnp'
  267 |     BleCompositeHIDInstance->_hid->pnp(vidSource, vid, pid, guidVersion);
      |                                    ^~~
/Users/laurinmichel/Documents/Arduino/libraries/ESP32-BLE-CompositeHID/BleCompositeHID.cpp:268:36: error: 'class NimBLEHIDDevice' has no member named 'hidInfo'; did you mean 'setHidInfo'?
  268 |     BleCompositeHIDInstance->_hid->hidInfo(0x00, 0x01);
      |                                    ^~~~~~~
      |                                    setHidInfo
/Users/laurinmichel/Documents/Arduino/libraries/ESP32-BLE-CompositeHID/BleCompositeHID.cpp:280:65: error: 'class NimBLEHIDDevice' has no member named 'hidService'; did you mean 'getHidService'?
  280 |     pAdvertising->addServiceUUID(BleCompositeHIDInstance->_hid->hidService()->getUUID());
[... cut ...]
   69 |     _mediaInput = hid->inputReport(MEDIA_KEYS_REPORT_ID);
      |                        ^~~~~~~~~~~
      |                        getInputReport
/Users/laurinmichel/Documents/Arduino/libraries/ESP32-BLE-CompositeHID/KeyboardDevice.cpp:70:20: error: 'class NimBLEHIDDevice' has no member named 'outputReport'; did you mean 'getOutputReport'?
   70 |     _output = hid->outputReport(_config.getReportId());
      |                    ^~~~~~~~~~~~
      |                    getOutputReport

exit status 1

Compilation error: exit status 1
```

cflixdev-projects, 2024-12-23:
The latest version of NimBLE includes additional features, which, despite containing the necessary commit, may still cause issues. To fix this, download version [1.4.0](https://github.com/h2zero/NimBLE-Arduino/tree/a79941cc9e77ed60136ee385b1c73d89c847bd94) as a ZIP file, extract it, and move the folder to your Arduino libraries directory (e.g., C:\Users\XXXXX\Documents\Arduino\libraries, or wherever your libraries are stored. You can also just install v1.4.0 in the Arduino Library.

Laurin08, 2024-12-23:
Hi
Thanks again, I was able to clear the error with adding `#include <stdexcept>` in the GamepadDevice.cpp file. now it works with the 2.0.0 version of nimble

Mystfit, 2024-12-27:
> Hi Thanks again, I was able to clear the error with adding `#include <stdexcept>` in the GamepadDevice.cpp file. now it works with the 2.0.0 version of nimble

I've merged in the latest NimBLE 2.1 support from #23. You should now be able to use the latest release version of NimBLE rather than relying on a specific commit.

closed 2025-03-05 by Mystfit

Laurin08, 2025-06-04:
Hi 
Ive got the same problem again, if i use " compositeHID.end();" the controller crashes....
do you have an idea what the problem could be?