Mystfit/ESP32-BLE-CompositeHID #26: <Callback.h> is missing

modest1981, 2025-01-06:
The file <Callback.h> is missing which results in below error during compile:

_In file included from C:\Users\Mohsen\AppData\Local\Temp\.arduinoIDE-unsaved202506-1532-1vdninh.wqz4\GamepadAndMouse\GamepadAndMouse.ino:4:
C:\Users\Mohsen\Documents\Arduino\libraries\ESP32-BLE-CompositeHID/GamepadDevice.h:7:10: fatal error: Callback.h: No such file or directory
    7 | #include <Callback.h>
      |          ^~~~~~~~~~~~
compilation terminated.
exit status 1

Compilation error: exit status 1_

LeeNX, 2025-01-06:
Have you installed the Arduino Dependencies ```Callback```? Available from the Arduino library manager or [Callbck](https://github.com/tomstewart89/Callback).

modest1981, 2025-01-06:
Oops! Thanks. I'm wondering why I couldn't find it by search.

LeeNX, 2025-01-07:
Can't say I have never done that myself. ;-)

If all works, please close. the issue. Thanks.

closed 2025-01-12 by modest1981