Mystfit/ESP32-BLE-CompositeHID #27: HID descriptor not updating

CasperHxR47, 2025-01-18:
Hello, 

I'm having trouble getting input through to my pc using this library. I think it may be a problem with my HID descriptor. 

Here is my code:
*******************************
#include <Arduino.h>
#include <BleConnectionStatus.h>
#include <BleCompositeHID.h>
#include <GamepadDevice.h>

#define button_A_PIN 32
#define button_B_PIN 33

BleCompositeHID compositeHID("Controller", "Me", 100);
GamepadDevice* gamepad;

void setup() {
    Serial.begin(115200);
    GamepadConfiguration bleGamepadConfig;
    bleGamepadConfig.setAutoReport(false);
    bleGamepadConfig.setControllerType(CONTROLLER_TYPE_GAMEPAD);
    bleGamepadConfig.setButtonCount(10);
    bleGamepadConfig.setHatSwitchCount(1);
    bleGamepadConfig.setWhichAxes(true,true,true,true,true,true,false,false);
    bleGamepadConfig.setWhichSimulationControls(false,false,false,false,false);
    bleGamepadConfig.setAxesMin(0x8001);
    bleGamepadConfig.setAxesMax(0x7FFF);

    gamepad = new GamepadDevice(bleGamepadConfig);
    compositeHID.addDevice(gamepad);
    compositeHID.begin();

    pinMode(button_A_PIN, INPUT_PULLUP);
    pinMode(button_B_PIN, INPUT_PULLUP);

    Serial.println("Controller Successfully Loaded!");
    delay(1000);
}

void loop() {

    if (compositeHID.isConnected()){

        bool button_A = digitalRead(button_A_PIN);
        if (!button_A) { 
            gamepad->press(BUTTON_1); 
            Serial.println("A"); 
            gamepad->sendGamepadReport(); 
        } else { 
            gamepad->release(BUTTON_1); 
            gamepad->sendGamepadReport(); 
        }

        bool button_B = digitalRead(button_B_PIN);
        if (!button_B) {
            gamepad->press(BUTTON_2); 
            Serial.println("B"); 
            gamepad->sendGamepadReport(); 
        } else { 
          gamepad->release(BUTTON_2); 
          gamepad->sendGamepadReport(); }
  }

    delay(20;
}

************************

Although my print statements work when the buttons are pressed, I'm not getting any input on the pc. The bluetooth connects just fine, but I noticed that even though I have my axis set to 6 and my buttons set at 10, the device on my pc when connected says there are 8 axis and 16 buttons. I've removed the device from my pc and reconnected via bluetooth, but same results. 

Any idea what I'm doing wrong?

Thank you.

CasperHxR47, 2025-01-18:
Apologies, my code pasted weird.

Mystfit, 2025-03-08:
I just tested this code on my own ESP32, it worked as expected. The number of buttons showed up as 10, and I was able to short pins 32 and 33 to ground which lit up buttons 1 and 2 in the joystick control panel. The only change I made was fixing a missing parenthesis from the delay on the second-to-last line.
[image]

Mystfit, 2025-03-10:
I'm going to mark this as completed for the moment, if you are still having problems then let me know.

closed 2025-03-10 by Mystfit