Mystfit/ESP32-BLE-CompositeHID #31: gamepad issue after disconnecting and reconnecting

lanshicai, 2025-03-03:
I have tried all the branches, only @ grimlakason/
After disconnecting and reconnecting, the ESP32-BLE-CompositeHID branch can continue to use the gamepad .

Land-Ey, 2025-03-05:
Got the same problem. :/
Just to tell you you're not alone. :)

Reconnect works fine, but controller buttons and axes won't get read anymore.

closed 2025-03-05 by Mystfit

lanshicai, 2025-03-06:
> Got the same problem. :/ Just to tell you you're not alone. :)
> 
> Reconnect works fine, but controller buttons and axes won't get read anymore.

Use this branch  ： https://github.com/Mystfit/ESP32-BLE-CompositeHID/issues/31#issuecomment-2700340591

Mystfit, 2025-03-06:
> > Got the same problem. :/ Just to tell you you're not alone. :)
> > Reconnect works fine, but controller buttons and axes won't get read anymore.
> 
> Use this branch ： [#31 (comment)](https://github.com/Mystfit/ESP32-BLE-CompositeHID/issues/31#issuecomment-2700340591)

I think that link is just to this issue. Did you mean to link to the master branch or something else?