Mystfit/ESP32-BLE-CompositeHID #32: 2 issues: XInput gamepad not working after reconnection / 16 buttons work AND don't work :)

Land-Ey, 2025-03-05:
Hi there,

I am a noob (former programmer, but new to C++), so please tread me like the noob I am.

Regarding the issues: BOTH occur if using XINPUT. I am working with an ESP32-S3.

1) I can pair the virtual Xbox controller without any problems, Windows' joy.cpl (and other tools like VPC Joystick Tester) show all buttons and axes working fine. But after a disconnect, the virtual controller still gets recognized by windows and gets connected - but windows doesn't detect any inputs anymore. I need to make Windows forget the controller, pair it again to make it work. I tried three different PCs, behavior is the same everywhere. I read about more users running into this problem - but all issue reports got marked as solved without mentioning any reason or remedy.

2) I found a way of sending not just 12 but 16 buttons, by defining additional buttons (e. g. sending 0x20 for button 14). This indeed does work - Windows' joy.cpl and other tools show the correct keypresses. But: The application I built my controller for (VPX virtual pinball) does not recognize the keystrokes in XInput and SDL mode, restricts the input to 12 buttons. Also Joy2Key does not see the additional buttons. I recently tried "AntiMicroX" - and that tool does the job. So I made it work - but I would love making it work without additional software in between.
DirectInput is no alternative as it doesn't support rumble - and I badly need the rumble signals for immersion.

Any ideas? - I just hope I made some typical noob mistakes. :D

Mystfit, 2025-03-05:
For issue 1, I've now started experiencing this myself with my XBoxXInputController sketch so I'll keep looking into this myself. Not quite sure what the problem is yet.

For issue 2, if you are using XInput, then the following may apply (taken from https://github.com/MHeironimus/ArduinoJoystickLibrary/wiki/FAQ). If you want to send more buttons then you'd need implement this as a generic gamepad

> XInput supports maximum of 4 axes, 10 buttons, 2 triggers and 8-direction digital pad per controller, compared to DirectInput's support for 8 axes, 128 buttons, and full-range POV. (The number of axes, buttons and triggers XInput supports corresponds directly to the Xbox 360 controller.)

Land-Ey, 2025-03-05:
Hi Mystfit, that was lightning-fast! :D

Regarding 1): Thank you very much for looking into it. 👍 

Regarding 2): The strange thing is that your solution DOES send the additional 4 buttons, Windows also can see the button presses. But most applications can't see them. Is there any workaround which does not sacrifice the rumble feature? Something like sending the button presses separately from the XInput part, maybe as keypresses?

My controller relies on the additional buttons for extra functionality like resetting the VR view or changing the height of a pinball table. I could use the D-Pad, but D-Pad also isn't recognized by the pinball application, just buttons and keys.
Yes, there's always something ... ;)

Mystfit, 2025-03-05:
If you would like keypresses, what about adding a KeyboardDevice to your BleCompositeHID instance after the XboxGamepadDevice is added? I'm not sure how this will play with XInput, but since it's technically a different part of the HID descriptor, maybe it will get passed along. See the MultipleDeviceExamples folder for some examples.

Windows picks the `Bluetooth LE XINPUT device` when we load up the XBoxXInputController sketch since we're passing along an XBox PID and VID value that Windows recognizes as an XBox controller. This loads both the "XInput" driver as well as a "HID Over GAT" driver, so the keyboard might spill over to the latter driver. Let me know if it works, and tomorrow I'll look into writing a new example to test if it works.

Mystfit, 2025-03-05:
> Thank you, I'll try accomplishing that KeyboardDevice thing - still
> fighting with C++ and it's environment.
>
> Regarding your latest commit: Strange thing - it did not work for the
> first and second time, still the same: flawless reconnect, but no
> buttons/axes reported. After pairing for the third time, it suddenly
> worked. I tried about 10 cycles, and every time I got readings after the
> reconnect.
> Will try tonight with my gaming PC in the natural habitat of my pinball
> controller and give you an update tomorrow.
> Thank you very much! :)

Let me know how it goes! Also, pinball controller? That sounds really
interesting! Are you controlling XInput with a pinball table?

Mystfit, 2025-03-08:
That looks so cool! I just tested the additional keyboard device idea and it seems to work. Steam still sees the ESP32 as an XBox controller, and I was able to receive keyboard inputs at the same time. I've uploaded a new [example ](https://github.com/Mystfit/ESP32-BLE-CompositeHID/blob/master/examples/MultipleDeviceExamples/XBoxControllerAndKeyboard/XBoxControllerAndKeyboard.ino) that will show you how to combine both devices together.

Closing the issue for the moment since I think we've solved the issues, but please feel free to submit any video or picture updates for your project to the [discussions forum](https://github.com/Mystfit/ESP32-BLE-CompositeHID/discussions/35).

closed 2025-03-08 by Mystfit