Mystfit/ESP32-BLE-CompositeHID #33: How do I set what pins are what buttons for Xinput?

MrDrTaos, 2025-03-05:
Hi, I'm an absolute noob on this so sorry if this is obvious.

I'm making a controller using an ESP 32 WROOM D and I want to use Xinput. I'm looking at the example of the controller and that's made a successful automated test program but I don't actually know how to say which pins are each xbox button. Any chance someone could help me out?

I know I need to define what the connected pins are but I don't know what to write where to make that happen.

Mystfit, 2025-03-06:
I'm assuming you mean that you're wanting to send XInput button presses when digital pins change their state?

You'll need to firstly determine when your digital pins change from LOW to HIGH or vice-versa. I recommend using a debounce library to filter out fluctuating button presses. When the state of your digital pin changes, you'll then either call `press(BUTTON_NAME)` or `release(BUTTON_NAME)`, where BUTTON_NAME is a valid XBOX_BUTTON value. See the `testButtons` function in `examples/GamepadExamples/XboxXInputController/XboxXInputController.ino` for how this is called.

I might look into writing another example that uses digital pins for buttons and analog pins for axes to make it a bit clearer.

LeeNX, 2025-03-10:
In my project [BLE Gamepad Collection](https://gitlab.com/leet/ble-gamepad-collection,) which is a work in progress, I have a good starting point [wiring-diagram](https://wokwi.com/projects/424305759534985217) and some test code, thou none yet for ```ESP32-BLE-CompositeHID``` nor any BLE code in [WokWi](https://wokwi.com), as they don't support Bluetooth stack.

It might be a good place to start with, without letting the magicSmoke out of any devices. Hope this helps somebody make any progress on their projects.

closed 2025-04-06 by Mystfit