Mystfit/ESP32-BLE-CompositeHID #37: ESP IDF port?

maxr217, 2025-03-27:
Hello, is there any potential to port this to an ESP IDF based library rather than just the arduino framework?

Mystfit, 2025-03-28:
I don't have plans for this currently but contributions are welcome. The NimBLE library is available for ESP IDF as well, so it should be possible to migrate it across. Most of the library uses stdlib variable types internally instead of arduino types (I think) so that might port across as well.

grimlokason, 2025-04-28:
Hello, 
you can use it with the ESP IDF !

maxr217, 2025-04-28:
Hello grimlokason,
Did you need to make any changes to import it? Did you just add the files as a custom component or something else?
Best,
Max

Mystfit, 2025-04-29:
> Hello, you can use it with the ESP IDF !

Oh really? Awesome! 

Did you have to include the Arduino component? I read that was something that needed to be included to get Arduino libraries working for ESP IDF.

JesseMelon, 2025-07-30:
I got it working no problem in ESP-IDF, I just had to make a bare minimum CMakeLists.txt, and the default compiler flags threw a couple errors which were < 5 minutes fix