Mystfit/ESP32-BLE-CompositeHID #38: [Requesting Help] Setting player indicator lights dynamically?

xanmankey, 2025-05-16:
First of all, awesome library, I've really enjoyed playing around with it the past couple days! I've been working on making a custom game controller using an ESP32-WROOM, and I saw that there was an example for setting player indicator lights. I was hoping to have LEDs that indicate the controller index, but I wasn't quite sure how to go about implementing this from the example. I setup the callback to indicate the controller index, but it never seems to run:

```c++
void OnPlayerIndicatorChanged(uint8_t playerIndicator)
{
  Serial.println("Player indicator changed to: " + String(playerIndicator));
  digitalWrite(LED_1, LOW);
  digitalWrite(LED_2, LOW);
  digitalWrite(LED_3, LOW);
  digitalWrite(LED_4, LOW);
  if (playerIndicator == 0) {
    digitalWrite(LED_1, HIGH);
  }
  else if (playerIndicator == 1) {
    digitalWrite(LED_1, HIGH);
    digitalWrite(LED_2, HIGH);
  }
  else if (playerIndicator == 2) {
    digitalWrite(LED_1, HIGH);
    digitalWrite(LED_2, HIGH);
    digitalWrite(LED_3, HIGH);
  }
  else if (playerIndicator == 3) {
    digitalWrite(LED_1, HIGH);
    digitalWrite(LED_2, HIGH);
    digitalWrite(LED_3, HIGH);
    digitalWrite(LED_4, HIGH);
  }
}

void setup()
{
   ...
  gamepad = new GamepadDevice(bleGamepadConfig); // Simulation controls, special buttons and hats 2/3/4 are disabled by default
  // Set LEDs based on controller index
  FunctionSlot<uint8_t> playerIndicatorChangedSlot(OnPlayerIndicatorChanged);
  gamepad->onPlayerIndicatorChanged.attach(playerIndicatorChangedSlot);

  compositeHID.addDevice(gamepad);
  compositeHID.begin();
}
```

I wasn't able to get the callback to trigger with the provided `PlayerIndicators.ino` example either, probably because I don't have a good grasp of what the actual trigger for the callback is. I'm new to HID report descriptors; from what I've read, it seems like the callback might be triggered by an output report descriptor issued by the host when controller indices change, but I'm not sure.

Any help you can provide is appreciated! Let me know if I can do anything to clarify.

Mystfit, 2025-05-19:
Hi @xanmankey , I wasn't able to ever test this properly as I wasn't able to find a way to set the player indicator data from the computer side in order to test the output reports. If you know of a way to set player LEDs on the host PC, then please let me know, as I can then test the output report alongside a normal controller.

LeeNX, 2025-05-19:
This is a area I am interested in and meant to reply over the weekend.  
  
If I understand the process, need to send commands to the Report(0x01), small python example here.

Install python ```hidapi``` with```pip install hidapi```

```python
import hid

def list_hid_devices():
    devices = hid.enumerate()
    if not devices:
        print("No HID devices found.")
        return None

    print("Available HID Devices:")
    for i, d in enumerate(devices):
        print(f"[{i}] {d['product_string']} - VID:PID {d['vendor_id']:04x}:{d['product_id']:04x}")
    return devices

def choose_device(devices):
    choice = int(input("Enter the number of the device to test: "))
    return devices[choice]

def test_output_report(device_info):
    try:
        dev = hid.Device(path=device_info['path'])

        # This is a sample output report to change Xbox 360 LED pattern
        # Format: Report ID (0x01), Command (0x03), Mode (e.g., 0x02 = LED 1 ON)
        report = bytes([0x01, 0x03, 0x02])

        print(f"Sending LED control report: {report}")
        bytes_written = dev.write(report)

        print(f"Bytes written: {bytes_written}")
        dev.close()
    except Exception as e:
        print(f"Failed to send report: {e}")

if __name__ == "__main__":
    devices = list_hid_devices()
    if devices:
        dev = choose_device(devices)
        test_output_report(dev)
```
I should test this on a Linux box, fails on this Mac.

Some info can be found at https://tattiebogle.net/ProjectRoot/Xbox360Controller/UsbInfo

Let us know if this works in any way. Maybe worth adding a working copy to a tools collection.