Mystfit/ESP32-BLE-CompositeHID #49: Two controllers acting as one

Gymcap, 2025-07-20:
I want to try to use this repo to make a split controller, like the joycon, where you have separate left and right controllers acting as one single controller. I've been trying to figure out what extra parts I'll need or what else I'll need to do to get two controllers working together like this. Would you be willing and able to tell me what I might need?

Mystfit, 2025-08-04:
I have a working version of a split controller using two ESP32s. It's actually the main inspiration for why I started using this library in the first place. I haven't worked on the project in a bit so it's not synced with the most recent library changes but take a look [here](https://github.com/Mystfit/Framework-Cyberdeck/tree/master/controller ) to see how I implemented it.

Mystfit, 2026-04-20:
Flagging this for myself to add ESP32 hub/spoke example using multiple controllers.